opengd/solar-sis #3: PIP4048 QPIGS2 command

cybersyx, 2018-05-17:
Hi, first of all, congratulations for the excellent work. 
I am using solar-sis with a PIP4048 inverter with two PV inputs. To take data from the second input, send the QPIGS2 command to the inverter. On the call.json I inserted a new query for the QPIGS2 command and aborted the query in the session.json file but I can not get any data. Perhaps the problem is that the CRC is calculated for a string (command) of 5 characters (QPIGS, QPIRI) while the new command that I have to send (QPIGS2) has 6 characters. How can you solve the problem? Thanks so much

kasperbirk, 2018-11-24:
Hello @daromer2 
I have extended the calls.json file with qpigs2 commands, and tried to push to GitHub, with no luck. The file is attached here. I haven't testet it yet, but will do soon..

[calls.json.txt](https://github.com/opengd/solar-sis/files/2612683/calls.json.txt)

cybertza, 2019-03-20:
so i have the issue with the qpigs2 command, that i dont know how to send the mode [l,b] to the database?

Here is my Calls.json
```

{
  "query_config": {
    "start_bit": "",
    "command_type": "P",
    "response_type": "D",
    "data_length_bits": 3,
    "seperator": " ",
    "ending_character": "\r",
    "crc_length": 2,
    "response_start": "(",
    "response_header_length": 0
  },
  "query": {
    "inverter1_status": {
      "command": "QPGS0",
      "influx": true,
      "cache": true,
      "crc16": true,
      "response": {
        "parallel_mode": 1,
		"serial_num": 1,
        "work_mode": false,
		"fualt_code": 1,
        "ac_in_volt": 1,
        "ac_in_freq": 1,
        "ac_out_volt": 1,
        "ac_out_freq": 1,
        "ac_out_power": 1,
        "ac_out_active_power": 1,
        "load_percent": 1,
		"battery_volt": 1,
        "battery_charging_current": 1,
        "battery_capacity": 1,
        "pv_input_voltage_1": 1,
        "total_charging_current": 1,
        "total_ac_output_apparent_power": 1,
		"total_ac_output_active_power": 1,
		"total_ac_output_percent": 1,
		"device_status": true,
		"output_mode": 1,
		"charger_source_priority": 1,
		"max_charge_current": 1,
		"max_charge_range": 1,
		"max_ac_charge_current": 1,
        "pv_input_current_for_battery": 1,
        "battery_discharge_current": 1
      }
    },
	"inverter2_status": {
      "command": "QPGS1",
      "influx": true,
      "cache": true,
      "crc16": true,
      "response": {
        "parallel_mode": 1,
		"serial_num": 1,
        "work_mode": false,
		"fualt_code": 1,
        "ac_in_volt": 1,
        "ac_in_freq": 1,
        "ac_out_volt": 1,
        "ac_out_freq": 1,
        "ac_out_power": 1,
        "ac_out_active_power": 1,
        "load_percent": 1,
		"battery_volt": 1,
        "battery_charging_current": 1,
        "battery_capacity": 1,
        "pv_input_voltage_1": 1,
        "total_charging_current": 1,
        "total_ac_output_apparent_power": 1,
		"total_ac_output_active_power": 1,
		"total_ac_output_percent": 1,
		"device_status": true,
		"output_mode": 1,
		"charger_source_priority": 1,
		"max_charge_current": 1,
		"max_charge_range": 1,
		"max_ac_charge_current": 1,
        "pv_input_current_for_battery": 1,
        "battery_discharge_current": 1
      }
    },
    "device_rated_information": {
      "command": "QPIRI",
      "influx": true,
      "cache": true,
      "crc16": true,
      "response": {
        "grid_rating_voltage": 1,
        "grid_rating_current": 1,
        "ac_output_rating_voltage": 1,
        "ac_output_rating_frecuency": 1,
        "ac_output_rating_current": 1,
        "ac_output_rating_apparent_power": 1,
        "ac_output_rating_active_power": 1,
        "battery_rating_voltage": 1,
        "battery_recharge_voltage": 1,
        "battery_under_voltage": 1,
        "battery_bulk_voltage": 1,
        "battery_float_voltage": 1,
        "battery_type": 1,
        "current_max_ac_charging": 1,
        "current_max_charging_current": 1,
        "input_voltage_range": 1,
        "output_source_priority": 1,
        "charger_source_priority": 1,
        "parallel_max_num": 1,
        "machine_type": 1,
        "topology": 1,
        "output_mode": 1,
        "battery_re_discharge_voltage": 1,
        "pv_condition": 1,
        "pv_power_balance": 1
      }
    }
  },
  "set_config": {
    "start_bit": "",
    "command_type": "S",
    "response_type": "",
    "data_length_bits": 0,
    "seperator": " ",
    "ending_character": "\r",
    "crc_length": 2,
    "response_accept_command": 1,
    "response_refuse_command": 0,
    "variable_length_fillout": "0",
    "response_start": "(",
    "response_header_length": 0
  },
  "set": {
    "feeding_grid_power_calibration":{
      "command": "S012FPADJ",
      "variables": {
        "feeding_grid_derection": [1, 0, 1],
        "feeding_grid_calibration_power": [4, 0, 999]
      }
    },
    "f1": {
      "command": "S013FP",
      "hide": true,
      "variables": {
        "fase": "",
        "watt": [4, 0, 999]
      }
    }
  }
}

```

and my session.json
```

{
  "serial_port": "/dev/hidraw0",
  "serial_baudrate": 2400,
  "serial_queue_delay": 200,
  "serial_parsers_readline": "\r",
  "serial_restart_threshold": 10000,
  "serial_clear_command_queue_on_restart": true,
  "http_port": 3001,
  "influx_pre_header": "pip_",
  "influxUrl": "http://localhost:8086/write?db=powerwall&precision=s",
  "IntervalCommands": [
    {
      "name": "inverter1_status",
      "config": "query_config",
      "command": "query/inverter1_status",
      "interval": 3000,
      "max": 0,
      "influx": true
    },
	{
      "name": "inverter2_status",
      "config": "query_config",
      "command": "query/inverter2_status",
      "interval": 3000,
      "max": 0,
      "influx": true
    },
    {
      "name": "device_rated_information",
      "config": "query_config",
      "command": "query/device_rated_information",
      "interval": 4000,
      "max": 0,
      "influx": true
    }
  ], 
  "OnInit": {
    "StartInterval": ["inverter1_status","inverter2_status", "device_rated_information"]
  }
}
```

kasperbirk, 2019-04-04:
Hello
Iam also trying to make QPIGS2 to work, but no luck yet?

For QPIGS2 I see this in the log:
[2019-04-04 07:40:35] info serial:/dev/hidraw0:SEND query/extended_status
[2019-04-04 07:40:35] info serial:/dev/hidraw0:SEND_RAW [{"type":"Buffer","data":[81,80,73,71,83,32,50,179]},{"type":"Buffer","data":[174,13]}]

And that differs from the general:
[2019-04-04 07:40:35] info serial:/dev/hidraw0:SEND query/general_status
[2019-04-04 07:40:35] info serial:/dev/hidraw0:SEND_RAW [{"type":"Buffer","data":[81,80,73,71,83,183,169,13]}]

The extended/qpigs2 has this part ekstra (,{"type":"Buffer","data":[174,13]})

Maybe it hasn't anything to do with my problem. My calls and session looks like this:

Calls
{
  "query_config": {
    "start_bit": "",
    "command_type": "P",
    "response_type": "D",
    "data_length_bits": 3,
    "seperator": " ",
    "ending_character": "\r",
    "crc_length": 2,
    "response_start": "(",
    "response_header_length": 0
  },
  "query": {
    "general_status": {
      "hide": false,
      "influx": true,
      "cache": true,
      "command": "QPIGS",
      "crc16": true,
      "response": {
        "grid_voltage": 1,
        "grid_frecuency": 1,
        "ac_output_voltage": 1,
        "ac_output_frecuency": 1,
        "ac_output_power_va": 1,
        "ac_output_active_power": 1,
        "output_load_percent": 1,
        "bus_voltage": 1,
        "battery_voltage": 1,
        "battery_charging_current": 1,
        "battery_capacity": 1,
        "inverter_heat_sink_temperature": 1,
        "pv_input_current_for_battery": 1,
        "pv_input_voltage_1": 1,
        "battery_voltage_from_scc": 1,
        "battery_discharge_current": 1,
        "device_status": true,
        "charging_scc": ["device_status", "8", "5", "6"],
        "charging_ac": ["device_status", "8", "6", "7"],
        "charging_scc_acc": ["device_status", "8", "7", "8"]
      }
    },
    "extended_status": {
     "hide": false,
      "influx": true,
      "cache": true,
      "command": "QPIGS 2", --> have also tried with: QPI¾¬
      "crc16": true,
        "response": {
        "pv_input_current_2": 1,
        "pv_input_voltage_2": 1,
        "battery_voltage_from_scc_2": 1,
        "pv_charging_power_2": 1,
        "ac_charging_current": 1,
        "ac_charging_power": 1,
        "pv_input_current_3": 1,
        "pv_input_voltage_3": 1,
        "battery_voltage_from_scc_3": 1,
        "pv_charging_power_3": 1,
        "pv_total_charging_power": 1
      }
    }
  }
}

Session
{
  "serial_port": "/dev/hidraw0",
  "serial_baudrate": 2400,
  "serial_queue_delay": 200,
  "serial_parsers_readline": "\r",
  "serial_restart_threshold": 10000,
  "serial_clear_command_queue_on_restart": true,
  "http_port": 8080,
  "influx_device": "pip",
  "influxUrl": "http://10.0.1.30:8086/write?db=pip4048mst&precision=s",
  "IntervalCommands": [
    {
      "name": "extended_status",
      "config": "query_config",
      "command": "query/extended_status",
      "interval": 5000,
      "max": 0,
      "influx": true
    },
    {
      "name": "general_status",
      "config": "query_config",
      "command": "query/general_status",
      "interval": 10000,
      "max": 0,
      "influx": true
    }
  ],

  "ListenOn": [],
  "OnInit": {
    "StartInterval": ["general_status","extended_status"]
  }
}

The output loks like this:
[2019-04-04 07:53:39] info serial:/dev/hidraw0:SEND query/extended_status
[2019-04-04 07:53:39] info serial:/dev/hidraw0:SEND_RAW [{"type":"Buffer","data":[81,80,73,190,172,36,20,13]}]
[2019-04-04 07:53:39] info serial:/dev/hidraw0:RECIVED_RAW Data: (NAKss Length: 8
[2019-04-04 07:53:39] info serial:/dev/hidraw0:RECIVED query/extended_status
device=pip,query=query_extended_status pv_input_current_2=NaN,pv_input_voltage_2=NaN,battery_voltage_from_scc_2=NaN,pv_charging_power_2=NaN,ac_charging_current=NaN,ac_charging_power=NaN,pv_input_current_3=NaN,pv_input_voltage_3=NaN,battery_voltage_from_scc_3=NaN,pv_charging_power_3=NaN,pv_total_charging_power=NaN
[2019-04-04 07:53:39] info influx:SEND solar,device=pip,query=query_extended_status pv_input_current_2=NaN,pv_input_voltage_2=NaN,battery_voltage_from_scc_2=NaN,pv_charging_power_2=NaN,ac_charging_current=NaN,ac_charging_power=NaN,pv_input_current_3=NaN,pv_input_voltage_3=NaN,battery_voltage_from_scc_3=NaN,pv_charging_power_3=NaN,pv_total_charging_power=NaN

@opengd @daromer2  - Can one of you help with this?

kasperbirk, 2019-04-04:
Got this from MPP solar:

CRC_README.txt

`QPI CRC value is: 0xBEAC

QPI: 0x51 0x50 0x49

Return: 0x0D

The QPI command (HEX value) sent to unit is:  515049BEAC0D`

CRC.c
`INT16U cal_crc_half(INT8U far *pin, INT8U len)
{

	INT16U crc;

	INT8U da;
	INT8U far *ptr;
	INT8U bCRCHign;
    INT8U bCRCLow;

	INT16U crc_ta[16]=
	{ 
		0x0000,0x1021,0x2042,0x3063,0x4084,0x50a5,0x60c6,0x70e7,

		0x8108,0x9129,0xa14a,0xb16b,0xc18c,0xd1ad,0xe1ce,0xf1ef
	};
	ptr=pin;
	crc=0;
	
	while(len--!=0) 
	{
		da=((INT8U)(crc>>8))>>4; 

		crc<<=4;

		crc^=crc_ta[da^(*ptr>>4)]; 

		da=((INT8U)(crc>>8))>>4; 

		crc<<=4;

		crc^=crc_ta[da^(*ptr&0x0f)]; 

		ptr++;
	}
	bCRCLow = crc;

    bCRCHign= (INT8U)(crc>>8);

	if(bCRCLow==0x28||bCRCLow==0x0d||bCRCLow==0x0a)

    {
    	bCRCLow++;
    }
    if(bCRCHign==0x28||bCRCHign==0x0d||bCRCHign==0x0a)

    {
		bCRCHign++;
    }
    crc = ((INT16U)bCRCHign)<<8;
    crc += bCRCLow;
	return(crc);
}
`

I also got the communication protocol but can't upload the pdf here..

cybertza, 2019-11-12:
#12 - This resolves the issue mentioned for PIP4048 in PAR mode.

daromer2, 2019-11-13:
Solved via #12

closed 2019-11-13 by daromer2