opengd/solar-sis #4: Error: Invalid argument setting custom baud rate of 2400

SilverNodashi, 2018-07-11:
I tried to run your code on a Raspberry Pi, connected to a PIP4048 inverter, on /dev/hidraw0 but got the following error:

`root@grafpi:/solar-sis/solar-sis/example/PIP4084# node project.js
[2018-07-11 18:06:05] http localhost Listening on port: 3001
[2018-07-11 18:06:05] ERR! serial:/dev/hidraw0 Error: Error: Invalid argument setting custom baud rate of 2400
[2018-07-11 18:06:05] ERR! serial:/dev/hidraw0     at Error (native)
[2018-07-11 18:06:05] ERR! serial:/dev/hidraw0  { Error: Error: Invalid argument setting custom baud rate of 2400
[2018-07-11 18:06:05] ERR! serial:/dev/hidraw0     at Error (native)
[2018-07-11 18:06:05] ERR! serial:/dev/hidraw0   stack: 'Error: Error: Invalid argument setting custom baud rate of 2400\n    at Error (native)' }
root@grafpi:/solar-sis/solar-sis/example/PIP4084# dmesg | grep hidraw
[    0.796647] hidraw: raw HID events driver (C) Jiri Kosina
[    2.259217] hid-generic 0003:0665:5161.0001: hiddev0,hidraw0: USB HID v1.11 Device [HID 0665:5161] on usb-3f980000.usb-1.3/input0
`

here's the history of commands I used:

` 320 mkdir /solar-sis
  321 cd /solar-sis
  321  sudo git clone https://github.com/opengd/solar-sis.git
  322  cd solar-sis/example/PIP4084
  323  ll
  324  more package.json
  325  more project.js
  326  ll
  327  more calls.json
  328  ll
  329  more calls.json
  330  ll
  331  more project.js
  332  kk
  333  ll
  334  more session.json
  335  sudo npm install --unsafe-perm
  336  node project.js

`

SilverNodashi, 2018-07-15:
Hi, 

Do you know what could cause this?

SilverNodashi, 2018-08-01:
Can someone please tell me what causes this error?

daromer2, 2018-08-01:
Above issue generally comes from a to new version of serial installed. We are working on an upgraded version but untill then you need to downgrade.

SilverNodashi, 2018-08-01:
Thank you. What serial are you referring to? nodejs serialport? How do I dowgrade it? I have googled, but couldn't find a suitable answer.

uploader18, 2018-09-02:
sudo npm install solar-sis@1.0.4 --unsafe-perm this verison worked for me and be sure that you stop solar-pip in webmin before you update or you will get an error

kasperbirk, 2018-11-27:
I keep getting this message when installing older solar-sis or older serialport version. Am i doing something wrong?
-----------------------------------------
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@8.12.0 (node-v57 ABI) (falling back to source compile with node-gyp) 
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-arm.tar.gz

uploader18, 2018-12-01:
you are using serial port newer than 4.0.7...first of all are you using daromers imagr or did you install serial port on your own and what versiin of raspberry pi do you have....do you use stretch version of raspbian?

daromer2, 2019-03-29:
Use the older serial port driver instead. This version of solar-sis have had issues with the latest. Please retest

closed 2019-03-29 by daromer2

jschaeke, 2019-05-04:
I have the same issue (on Dietpi , stretch). Downgraded node to lts/carbon, tried various serialport versions between 3.1.2 and latest. What specific version should it work with or any other solutions? (https://github.com/nrm21/skymax-demo works on /dev/hidraw0, but I'd rather use this project).

jschaeke, 2019-05-04:
Found it, I had to update it in node_modules/solar-sis itself.
```/solar-sis/example/5048GK/node_modules/solar-sis# npm i serialport@4.0.7```

cowpen, 2020-02-20:
I'm trying to get solar-sis working on a RPi 3 B+. I'm unable to downgrade to serialport@1.0.4. Getting messages that it's deprecated, unavailable, EOL, etc. Is there a source and a manual method to install 1.0.4?

> gyp ERR! cwd /home/pi/solar-sis/example/PIP4084/node_modules/serialport
gyp ERR! node -v v10.19.0
gyp ERR! node-gyp -v v5.0.7
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1

jschaeke, 2020-04-04:
Yeah I tried it now on another rpi, can't get it running anymore either...

grand248, 2021-03-24:
I'm struggling to install serialport@4.0.7 on a R4, looks like download is no longer available:
```
pi@raspberrypi:~/solar-sis/example/PIP4084/node_modules/solar-sis $ npm i serialport@4.0.7
npm WARN npm npm does not support Node.js v10.24.0
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/

> serialport@4.0.7 install /home/pi/solar-sis/example/PIP4084/node_modules/solar-sis/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/usr/bin/node',
node-pre-gyp verb cli   '/home/pi/solar-sis/example/PIP4084/node_modules/solar-sis/node_modules/serialport/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.32
node-pre-gyp info using node@10.24.0 | linux | arm
node-pre-gyp verb command install []
node-pre-gyp info check checked for "/home/pi/solar-sis/example/PIP4084/node_modules/solar-sis/node_modules/serialport/build/Release/serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-arm.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-arm.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@10.24.0 (node-v64 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-arm.tar.gz
node-pre-gyp verb command build [ 'rebuild' ]
node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@10.24.0 (node-v64 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http Connection closed while downloading tarball file
node-pre-gyp verb command build [ 'rebuild' ]
```

b4dm00n, 2021-06-05:
I when i use "npm i serialport@4.0.7" i got this error . any 1 know why ?

> pi@raspberrypi:~/solar-sis/example/PIP4084/node_modules/solar-sis $ sudo npm i serialport@4.0.7
> npm ERR! code 1
> npm ERR! path /home/pi/solar-sis/example/PIP4084/node_modules/solar-sis/node_modules/serialport
> npm ERR! command failed
> npm ERR! command sh -c node-pre-gyp install --fallback-to-build
> npm ERR! make: Entering directory '/home/pi/solar-sis/example/PIP4084/node_modules/solar-sis/node_modules/serialport/build'
> npm ERR!   CXX(target) Release/obj.target/serialport/src/serialport.o
> npm ERR! make: Entering directory '/home/pi/solar-sis/example/PIP4084/node_modules/solar-sis/node_modules/serialport/build'
> npm ERR!   CXX(target) Release/obj.target/serialport/src/serialport.o
> npm ERR! make: Leaving directory '/home/pi/solar-sis/example/PIP4084/node_modules/solar-sis/node_modules/serialport/build'
> npm ERR! Failed to execute '/usr/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/solar-sis/example/PIP4084/node_modules/solar-sis/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/solar-sis/example/PIP4084/node_modules/solar-sis/node_modules/serialport/build/Release' (1)
> npm ERR! make: Leaving directory '/home/pi/solar-sis/example/PIP4084/node_modules/solar-sis/node_modules/serialport/build'
> npm ERR! node-pre-gyp info it worked if it ends with ok
> npm ERR! node-pre-gyp info using node-pre-gyp@0.6.32
> npm ERR! node-pre-gyp info using node@10.24.0 | linux | arm

cybertza, 2021-06-05:
There is a solution somewhere online to install that ver of serial, the newer Vers don’t work I believe

cybertza, 2021-06-05:
I think I managed to yarn that ver if I recall but it’s been some time