opengd/solar-sis #6: i can't build

Tolmino, 2019-03-22:
hi, i'm using raspbian 2018-11-13 
node v10.15.3
npm 6.9.0

during installation i have:

./src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:369:32: error: no matching function for call to ‘v8::Value::ToInt32()’
   baton->fd = info[0]->ToInt32()->Int32Value();

../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:480:29: error: no matching function for call to ‘v8::Value::ToInt32()’
   int fd = info[0]->ToInt32()->Int32Value();

../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:528:29: error: no matching function for call to ‘v8::Value::ToInt32()’
   int fd = info[0]->ToInt32()->Int32Value();
.
.
.
n file included from ../src/./serialport.h:7:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
serialport.target.mk:101: recipe for target 'Release/obj.target/serialport/src/serialport.o' failed
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/home/pi/solar-sis.old/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.14.98-v7+

cybertza, 2020-03-20:
ha@power:~$ node -v
v8.10.0
ha@power:~$ npm -v
3.5.2

i don't recall building, i just did a npm install from the base, and it worked, make sure you have make,gcc and all those installed. let me know

Tolmino, 2020-05-28:
found a problem with version of serial.
here i have resolved simply copying working node modules from old solar-sis version directory to new.

closed 2020-11-29 by daromer2