opengd/solar-sis #7: Change charge voltage based on SOC

SilverNodashi, 2019-03-29:
Hi, 
It's great that you now have support for the Pylontech batteries. 
Would it possibly to adjust the charge and discharge voltages on the various inverters based on the SOC of the Pylontech?

The PIP4048 sees higher voltages than what the battery actually has so it switches off before the battery has reached 80% DOD

daromer2, 2019-03-29:
This has nothing to do with either Pylontech nor the inverter it self. Solar-sis is just a software that makes it easier to call serial interfaces on a charger/inverter by serialising the protocol kind of.  The Voltage vs SOC is not something you set in this software but on your inverter.  Note that the PIP do not support anything but LA so those numbers wont work

closed 2019-03-29 by daromer2

SilverNodashi, 2019-03-29:
I hear you but that means I have to physically be at the invert to make the changes, which I'm not. 
Is there no way to send commands to the PIP inverter from solarsis?

daromer2, 2019-03-29:
As I said the solar-sis have nothing to do with calibration of the inverter or the hardware itself. You seem to have an issue related to your setup if your voltage is that much of. And the SOC vs Voltage is in the software of the inverter. 
Solar-sis is just a tool to read data from its serial interface and parse it into MQTT or Influx data.

SilverNodashi, 2019-03-29:
Daniel. I don't have an issue. I am proposing making the system work a bit
better. Graphing is nice, but it would be better if solarsis can control
the inverter as well.
The inverter can take commands, to for example switch from solar to grid
feed, and it would be great, as example, to switch back to grid from
batteries if solar-sis sees that the battery bank is getting close to
empty, to avoid sitting in the dark while I'm sleeping and not being able
to switch it manually. But for this to work properly, the inverter needs to
have the right soc readings.

This is just one usage scenario. I monitor our environment as well and I
would like to be able to switch back to grid when it's too cloudy, or
rainy, for example. I am not always home so I cannot just quickly change
the settings on the inverter in such a case.

See here for more info:
http://forums.aeva.asn.au/viewtopic.php?p=71631#p71631

daromer2, 2019-03-29:
It already can do that so Im not sure what you look for? solar-sis is not a logic-software and will never be. 

 I control my inverters via solar-sis but solar-sis is not the brain. Solar sis is only made to be a middle point to support sending and interpretation of commands. I'm using a 3rd party script to hook all my systems together for the fine part but first of all I use my inverters to automatically switch when needed. I have set their limits to work and for all fine tuning like "no sun for 5 days-prediction" I let the 3rd party software to switch earlier. 

You can add all commands you want in the config and you are free to submit updates here and I incorporate them. 

In my case i have some logic in node-red to controll emergency settings needed as last resort.

Hope that answers your thoughts.

SilverNodashi, 2019-03-29:
I don't know NodeJS nor can't code it :(

daromer2, 2019-03-29:
Dont have to  :)  I use Node-Red as one of the tools and thats visual where you draw flows. I use solar-sis MQTT and the Web api on it to control stuff.

cybertza, 2019-04-27:
> Dont have to :) I use Node-Red as one of the tools and thats visual where you draw flows. I use solar-sis MQTT and the Web api on it to control stuff.

Where can i read more about the web API or MQTT commands to set inverter values?

cybertza, 2019-04-27:
https://secondlifestorage.com/t-Esperyd%C2%B4s-Raspberry-PI-project?pid=33122#pid33122