opengd/solar-sis #9: Wiring of Us2000B

SilverNodashi, 2019-04-25:
Hi, 

Would you mind telling me which port you use on the US2000B battery, along with the code that's included in solar-sis for the US2000B? I suspect I you might be connecting to the Serial Console on the US2000B? And if so, do you connect it directly to the Raspberry Pi's Serial Port, or do you use a USB to Serial converter?

cybertza, 2019-11-12:
Hey, Ive just done this

You connect to serial port/serial port to usb, both should word, just update your session.js for the correct serial port, and it works.

I also had to tweak the calls for the amount of packs i had before it would submit to Influx.

cybertza, 2020-03-20:
i use the console port with a rs232 device, has been working great for a while, there is some guides online to make the cable, easiest way to find them is to look at the firmware upgrade video/guide.

let me know if this is good?

closed 2020-11-29 by daromer2