opengd/solar-sis #11: implentation of pip5048mk watt output from pv

affer, 2019-09-04:
in my output from terminal:
[2019-09-04 08:04:29] info serial:/dev/hidraw0:RECIVED_RAW Data: (220.2 49.9 230.0 50.0 0207 0151 004 335 48.70 000 010 0029 0001 069.7 48.62 00002 00010110 00 00 00068 010� Length: 109
i get the last numbers in the string: "00068" which is my pv output, could that be implemented to pv_input_watt so it can be used to present in grafana?

daromer2, 2020-01-21:
You can submit a fix for that. But im not sure its included on all pip systems., Perhaps only the new one?

Josuan, 2020-02-21:
the way i do it is, field(pv_input_current_for_battery) mean() math (*mean(pv_input_voltage_1))

cybertza, 2020-02-21:
yup, that is what i used to implement as well, programatically i don't know if its worth the effort if it can be done on the db side.

Josuan, 2020-04-21:
using pip4048 example, in a 5048gk model the query pv_input_current_for_battery, represent amps coming from pv, but in a 5048mg the same query represent amps going to battery bank, so in a 5048mg i cant do the math like above to obtain pv watts

cybertza, 2020-04-21:
you can perhaps check what command does give you the solar watts in the proto doc thats is around:
if you let me know perhaps i can help you to update for mg, somehow i remember there was an http://forums.aeva.asn.au/viewtopic.php?t=4332 I have eventually upgraded to a SUN-SYNK-8K inverter, so i need to learn how modbus works, but the lockdown has prevented me on getting my hands on 485 devices.

Tolmino, 2020-05-03:
hi, can someone give me the right query? i'm trying to receive RAW ,
but i have only that logs:
[2020-05-03 10:41:07] http localhost Listening on port: 3001
[2020-05-03 10:41:07] info serial:/dev/hidraw0 Open serial port
[2020-05-03 10:41:07] info queue:ADD_INTERVAL query/general_status:QPIGS·©
[2020-05-03 10:41:07] info queue:ADD_INTERVAL query/device_rated_information:QPIRIøT
[2020-05-03 10:41:10] info serial:/dev/hidraw0:SEND query/general_status
[2020-05-03 10:41:10] info serial:/dev/hidraw0:SEND_RAW [{"type":"Buffer","data":[81,80,73,71,83,183]},{"type":"Buffer","data":[169,13]}]
[2020-05-03 10:41:10] info serial:/dev/hidraw0:RECIVED_RAW Data: (NAKss Length: 6
[2020-05-03 10:41:10] info serial:/dev/hidraw0:RECIVED query/general_status
[2020-05-03 10:41:10] info influx:SEND undefinedquery_general_status grid_voltage=NaN,grid_frecuency=NaN,ac_output_voltage=NaN,ac_output_frecuency=NaN,ac_output_power_va=NaN,ac_output_active_power=NaN,output_load_percent=NaN,bus_voltage=NaN,battery_voltage=NaN,battery_charging_current=NaN,battery_capacity=NaN,inverter_heat_sink_temperature=NaN,pv_input_current_for_battery=NaN,pv_input_voltage_1=NaN,battery_voltage_from_scc=NaN,battery_discharge_current=NaN,Battery voltage offset for fans on=NaN,EEPROM version=NaN,PV Charging power=NaN,Device Status=NaN

Tolmino, 2020-05-03:
my call.json:

{
  "query_config": {
    "start_bit": "",
    "command_type": "P",
    "response_type": "D",
    "data_length_bits": 3,
    "seperator": " ",
    "ending_character": "\r",
    "crc_length": 2,
    "response_start": "(",
    "response_header_length": 0
  },
  "query": {
    "general_status": {
      "hide": false,
      "influx": true,
      "cache": true,
      "command": "QPIGS",
      "crc16": true,
      "response": {
        "grid_voltage": 1,
        "grid_frecuency": 1,
        "ac_output_voltage": 1,
        "ac_output_frecuency": 1,
        "ac_output_power_va": 1,
        "ac_output_active_power": 1,
        "output_load_percent": 1,
        "bus_voltage": 1,
        "battery_voltage": 1,
        "battery_charging_current": 1,
        "battery_capacity": 1,
        "inverter_heat_sink_temperature": 1,
        "pv_input_current_for_battery": 1,
        "pv_input_voltage_1": 1,
        "battery_voltage_from_scc": 1,
        "battery_discharge_current": 1,
        "device_status": true,
        "Battery_voltage_offset_for_fans_on": 1,
        "EEPROM_version": 1,
        "PV_Charging_power": 1,
        "Device_Status": 1
      }
    },
    "device_rated_information": {
      "command": "QPIRI",
      "influx": true,
      "cache": true,
      "crc16": true,
      "response": {
        "grid_rating_voltage": 1,
        "grid_rating_current": 1,
        "ac_output_rating_voltage": 1,
        "ac_output_rating_frecuency": 1,
        "ac_output_rating_current": 1,
        "ac_output_rating_apparent_power": 1,
        "ac_output_rating_active_power": 1,
        "battery_rating_voltage": 1,
        "battery_recharge_voltage": 1,
        "battery_under_voltage": 1,
        "battery_bulk_voltage": 1,
        "battery_float_voltage": 1,
        "battery_type": 1,
        "current_max_ac_charging": 1,
        "current_max_charging_current": 1,
        "input_voltage_range": 1,
        "output_source_priority": 1,
        "charger_source_priority": 1,
        "parallel_max_num": 1,
        "machine_type": 1,
        "topology": 1,
        "output_mode": 1,
        "battery_re_discharge_voltage": 1,
        "pv_condition": 1,
        "pv_power_balance": 1,
        "Max_charging_time_at_C.V_stage": 1,
	"Operation_Logic": 1
      }
    }
  }

closed 2020-11-29 by daromer2