opengd/solar-sis #21: query daily solar and ac out kwh

Josuan, 2020-02-23:
can a query of daily solar kwh production and inverter kwh be done in session.json? thanks in advance

daromer2, 2020-02-24:
Depends on what protocol is used and IF the inverter you have supporta the 
value. IF not use influxdb

cybertza, 2020-02-24:
Do you have the Q for influxdb for that? i was considering rather using emon cms to display.

Josuan, 2020-02-24:
im using the pip4048 example folder, my inverter is a 5048gk

cybertza, 2020-03-20:
You can also write the values to EmonCMS and get the values from there, i found it a bit easier when i was using this.

daromer2, 2020-11-29:
to get a total number of something use integral in influxdb. Ie calculate the area of something. Thats the easiest if the inverter doesnt support it.

closed 2020-11-29 by daromer2