opengd/solar-sis #22: get =NaN values on hybrid v2

Josuan, 2020-03-07:
hi when i try to use solar-sis on a 5048 v2 i get all values from query =NaN, tried with the MPI10kWHybrid example as well with no avail, thanks in advance.

[2020-03-07 01:50:22] info serial:/dev/ttyUSB0:SEND query/general_status
[2020-03-07 01:50:22] info serial:/dev/ttyUSB0:SEND_RAW [{"type":"Buffer","data":[81,80,73,71,83,183,169,13]}]
[2020-03-07 01:50:22] info serial:/dev/ttyUSB0:RECIVED_RAW Data: �^0ength: 4
[2020-03-07 01:50:22] info serial:/dev/ttyUSB0:RECIVED query/general_status
device=pip,query=query_general_status grid_voltage=NaN,grid_frecuency=NaN,ac_output_voltage=NaN,ac_output_frecuency=NaN,ac_output_power_va=NaN,ac_output_active_power=NaN,output_load_percent=NaN,bus_voltage=NaN,battery_voltage=NaN,battery_charging_current=NaN,battery_capacity=NaN,inverter_heat_sink_temperature=NaN,pv_input_current_for_battery=NaN,pv_input_voltage_1=NaN,battery_voltage_from_scc=NaN,battery_discharge_current=NaN
[2020-03-07 01:50:22] info influx:SEND solar,device=pip,query=query_general_status grid_voltage=NaN,grid_frecuency=NaN,ac_output_voltage=NaN,ac_output_frecuency=NaN,ac_output_power_va=NaN,ac_output_active_power=NaN,output_load_percent=NaN,bus_voltage=NaN,battery_voltage=NaN,battery_charging_current=NaN,battery_capacity=NaN,inverter_heat_sink_temperature=NaN,pv_input_current_for_battery=NaN,pv_input_voltage_1=NaN,battery_voltage_from_scc=NaN,battery_discharge_current=NaN

cybertza, 2020-03-12:
info serial:/dev/ttyUSB0:RECIVED_RAW Data: �^0ength: 4

problem is that you getting bad data back, so the decode doesnt work, may find that your command out is incorrect

closed 2020-03-17 by Josuan

Tolmino, 2020-05-03:
someone has right command ?