opengd/solar-sis #23: cant get the mqtt option to work

Josuan, 2020-03-17:
i have home assistant installed and the mqtt addon, i can see the ip of the raspberry pi that solar-sis is installed and running on the log as connected but dont know how to add it or sub to a topic like battery_voltage

cybertza, 2020-03-17:
My latest PR had MQTT activated. I don’t know how it was implemented before, I just write every value to mqtt so you would just sub to what you need, or use node red or hass to calc if that’s required I have migrated over to solpiplog since it has a GUI and so
Before I had MQTT enabled on solar sis I was using node red to pull the last value from influx dB, and then writing it to a MQTT topic for hass. Was working fine as well. You can also do commands via the sol sis http port, had that working rather well also, from hass.io -> MQTT then MQTT to http with node red to parse it.

I still have some of the code around if you need.

Regards

Jaco Goussard

Josuan, 2020-03-19:
hi cybertza thanks for your reply if you could post your node-red flow so i can import it that would be awesome to try, thanks in advance

cybertza, 2020-03-20:
Hi

Sorry i dont have the full one i used to use anymore,
But here is an EG that can help you get there.

`[{"id":"86cd1122.6012","type":"mqtt out","z":"a4af1532.59d5e8","name":"","topic":"Batrium/SOC","qos":"","retain":"","broker":"d38b37e2.c1ab18","x":2510,"y":1060,"wires":[]},{"id":"c4c0f1e.10f2d1","type":"mqtt out","z":"a4af1532.59d5e8","name":"","topic":"Batrium/ShuntVoltage","qos":"","retain":"","broker":"d38b37e2.c1ab18","x":2540,"y":1200,"wires":[]},{"id":"c877528b.9c4ba","type":"mqtt out","z":"a4af1532.59d5e8","name":"","topic":"Batrium/ShuntCurrent","qos":"","retain":"","broker":"d38b37e2.c1ab18","x":2540,"y":1240,"wires":[]},{"id":"3bf8ab5d.5c9a84","type":"delay","z":"a4af1532.59d5e8","name":"","pauseType":"rate","timeout":"5","timeoutUnits":"seconds","rate":"1","nbRateUnits":"5","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":true,"x":2320,"y":1200,"wires":[["c4c0f1e.10f2d1"]]},{"id":"eaa4f51c.b12df8","type":"delay","z":"a4af1532.59d5e8","name":"","pauseType":"rate","timeout":"5","timeoutUnits":"seconds","rate":"1","nbRateUnits":"5","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":true,"x":2320,"y":1240,"wires":[["c877528b.9c4ba"]]},{"id":"cbd322bd.b900e","type":"influxdb in","z":"a4af1532.59d5e8","influxdb":"5cca6087.b8801","name":"Pylon","query":"SELECT last(*) FROM \"pylon\"","rawOutput":false,"precision":"","retentionPolicy":"","x":1850,"y":1080,"wires":[["8fbc68fd.49eaf8","70ce6918.1f1dc8","14106cdc.bad0f3"]]},{"id":"8fbc68fd.49eaf8","type":"function","z":"a4af1532.59d5e8","name":"","func":"msg.payload = msg.payload[0].last_Pack_voltage / 1000;\nreturn msg;","outputs":1,"noerr":0,"x":2110,"y":1200,"wires":[["3bf8ab5d.5c9a84"]]},{"id":"70ce6918.1f1dc8","type":"function","z":"a4af1532.59d5e8","name":"","func":"msg.payload = ((msg.payload[0].last_Pack_remains_mAh + msg.payload[0].last_Pack_remains_mAh_P2) / (msg.payload[0].last_Pack_total_mAh + msg.payload[0].last_Pack_total_mAh_P2 ));\nmsg.payload = msg.payload * 100;\nmsg.payload = Math.round(msg.payload);\nreturn msg;","outputs":1,"noerr":0,"x":2110,"y":1080,"wires":[["86cd1122.6012"]]},{"id":"14106cdc.bad0f3","type":"function","z":"a4af1532.59d5e8","name":"","func":"msg.payload = msg.payload[0].last_Pack_current / 10;\nreturn msg;","outputs":1,"noerr":0,"x":2110,"y":1240,"wires":[["eaa4f51c.b12df8"]]},{"id":"d38b37e2.c1ab18","type":"mqtt-broker","z":"","name":"","broker":"172.16.103.9","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closePayload":"","willTopic":"","willQos":"0","willPayload":""},{"id":"5cca6087.b8801","type":"influxdb","z":"a4af1532.59d5e8","hostname":"172.16.103.9","port":"8086","protocol":"http","database":"powerwall","name":"","usetls":false,"tls":""}]`

cybertza, 2020-03-20:
just play around a bit with the data coming in, and so forth, if you don't know node red, you will have a good understanding after this.

you would need the influx db palette for node.

cybertza, 2020-03-20:
Here is a control EG as well.

`[{"id":"bd886be7.199e68","type":"http request","z":"14ecd37c.c379ed","name":"Make Request To Inverter","method":"GET","ret":"txt","paytoqs":false,"url":"","tls":"","proxy":"","authType":"","x":930,"y":560,"wires":[["9b720545.9df338","4646e1a0.dedd1"]]},{"id":"bd204bc4.aaa4f8","type":"inject","z":"14ecd37c.c379ed","name":"","topic":"","payload":"57.54","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":230,"y":500,"wires":[["5cee94d8.8ae8dc"]]},{"id":"bdd06f37.da03c","type":"inject","z":"14ecd37c.c379ed","name":"","topic":"","payload":"57.54","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":230,"y":540,"wires":[["fcbd6351.c9961"]]},{"id":"56b6aa03.49cd74","type":"inject","z":"14ecd37c.c379ed","name":"","topic":"","payload":"48","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":230,"y":580,"wires":[["247e1a84.a726d6"]]},{"id":"7d82692e.4c3108","type":"inject","z":"14ecd37c.c379ed","name":"","topic":"","payload":"55","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":230,"y":620,"wires":[["fb86792.3522788"]]},{"id":"5cee94d8.8ae8dc","type":"function","z":"14ecd37c.c379ed","name":"Set Bulk Charge","func":"msg.url = \"http://172.16.103.28:3001/set/bulk_charging?voltage=\" + msg.payload;\nreturn msg;","outputs":1,"noerr":0,"x":400,"y":500,"wires":[["45e58317.415bcc"]]},{"id":"fcbd6351.c9961","type":"function","z":"14ecd37c.c379ed","name":"Set Float Charge","func":"msg.url = \"http://172.16.103.28:3001/set/float_charging?voltage=\" + msg.payload;\nreturn msg;","outputs":1,"noerr":0,"x":410,"y":540,"wires":[["45e58317.415bcc"]]},{"id":"247e1a84.a726d6","type":"function","z":"14ecd37c.c379ed","name":"Set Back To Grid","func":"msg.url = \"http://172.16.103.28:3001/set/back_to_grid=\" + msg.payload;\nreturn msg;","outputs":1,"noerr":0,"x":410,"y":580,"wires":[["45e58317.415bcc"]]},{"id":"fb86792.3522788","type":"function","z":"14ecd37c.c379ed","name":"Back to Batery","func":"msg.url = \"http://172.16.103.28:3001/set/back_to_battery?voltage=\" + msg.payload;\nreturn msg;","outputs":1,"noerr":0,"x":400,"y":620,"wires":[["45e58317.415bcc"]]},{"id":"45e58317.415bcc","type":"delay","z":"14ecd37c.c379ed","name":"","pauseType":"rate","timeout":"2","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":730,"y":560,"wires":[["bd886be7.199e68"]]}]`

cybertza, 2020-03-20:
these were working with the latest build i made, for Dual pip, you may have to adjust accordingly to your calls file.

Josuan, 2020-03-20:
hi cybertza thanks for your time and example! got it imported and modified the influx, function and mqtt nodes, :) now i got stuck on how to edit my configuration yml to add it under sensors: if you dont mind write a little example i would appreciate it very much thanks!

Josuan, 2020-03-20:
this is a flow i have to get the values from a sonoff s31 and populating a influxdb, i would like to do the opposite, read from a influxdb and send data to a broker topic and display it in ha

- platform: mqtt
    name: "Washer Voltage"
    state_topic: "tele/Washer/SENSOR"
    value_template: '{{value_json["ENERGY"]["Voltage"] }}'
    qos: 2
    unit_of_measurement: 'V'
    icon: mdi:flash-circle
    availability_topic: "tele/Washer/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"

[{"id":"1f187af9.f646c5","type":"mqtt in","z":"32901379.04e97c","name":"Fridge","topic":"tele/Fridge/SENSOR","qos":"2","datatype":"auto","broker":"","x":410,"y":660,"wires":[["9208302b.4b169"]]},{"id":"9208302b.4b169","type":"json","z":"32901379.04e97c","name":"ENERGY","property":"payload","action":"","pretty":true,"x":560,"y":660,"wires":[["3bc1f56e.68302a","bbfa7224.abcca"]]},{"id":"3bc1f56e.68302a","type":"function","z":"32901379.04e97c","name":"function","func":"if (msg.payload.ENERGY) {\n    var Power = { payload: msg.payload.ENERGY.Power, topic: \"Power\"  };\n    \nreturn [ Power ,];\n}","outputs":1,"noerr":0,"x":720,"y":660,"wires":[["9d48d14.d4c043"]]},{"id":"bbfa7224.abcca","type":"debug","z":"32901379.04e97c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":true,"complete":"payload","targetType":"msg","x":730,"y":580,"wires":[]},{"id":"9d48d14.d4c043","type":"influxdb out","z":"32901379.04e97c","influxdb":"","name":"write Power","measurement":"Power","precision":"","retentionPolicy":"","x":910,"y":660,"wires":[]}]

cybertza, 2020-03-20:
` - platform: mqtt
   name: "BVM AH Consumed"
   state_topic: "BMV/AH_Consumed"
 - platform: mqtt
   name: "BVM Amps"
   state_topic: "BMV/Amps"
 - platform: mqtt
   name: "BVM Amps"
   state_topic: "BMV/Amps"
 - platform: mqtt
   name: "BMV State Of Charge"
   state_topic: "BMV/SOC"
 - platform: mqtt
   name: "BMV Voltage"
   state_topic: "BMV/Voltage"
 - platform: mqtt
   name: "BMV Watts"
   state_topic: "BMV/Watts"
 - platform: mqtt
   name: "BMV Time To Go"
   state_topic: "BMV/TimeToGo"
 - platform: mqtt
   name: "BMV Time Full"
   state_topic: "BMV/TimeTo100"
 - platform: mqtt
   name: "Inverter SOC"
   state_topic: "Inverter/BatterySOC"
 - platform: mqtt
   name: "Inverter Volts"
   state_topic: "Inverter/BatteryVolts"`

cybertza, 2020-03-20:
- platform: mqtt
   name: "Inverter 2 Solar"
   state_topic: "tele/inverter2/pvwatts"
   unit_of_measurement: "W"

cybertza, 2020-03-20:
those are in my sensor.yaml file or under the sensor heading if you don't use files.

Josuan, 2020-03-20:
got it to work! thank you for your time much appreciated!

closed 2020-03-20 by Josuan