opengd/solar-sis #25: pip 5048 MK

Tolmino, 2020-05-27:
hi, when i'm using solar-sis on pip5048MK i have:
root@raspberrypi:/home/pi/solar-sis/example/PIP4084# node project.js 
[2020-05-27 12:14:11] http localhost Listening on port: 3001
[2020-05-27 12:14:12] info serial:/dev/hidraw0 Open serial port
[2020-05-27 12:14:12] info queue:ADD_INTERVAL query/general_status:QPIGS·©
[2020-05-27 12:14:12] info queue:ADD_INTERVAL query/device_rated_information:QPIRIøT
[2020-05-27 12:14:15] info serial:/dev/hidraw0:SEND query/general_status
[2020-05-27 12:14:15] info serial:/dev/hidraw0:SEND_RAW [{"type":"Buffer","data":[81,80,73,71,83,183]},{"type":"Buffer","data":[169,13]}]
[2020-05-27 12:14:15] info serial:/dev/hidraw0:RECIVED_RAW Data: (NAKss Length: 6
[2020-05-27 12:14:15] info serial:/dev/hidraw0:RECIVED query/general_status
[2020-05-27 12:14:15] info influx:SEND undefinedquery_general_status grid_voltage=NaN,grid_frecuency=NaN,ac_output_voltage=NaN,ac_output_frecuency=NaN,ac_output_power_va=NaN,ac_output_active_power=NaN,output_load_percent=NaN,bus_voltage=NaN,battery_voltage=NaN,battery_charging_current=NaN,battery_capacity=NaN,inverter_heat_sink_temperature=NaN,pv_input_current_for_battery=NaN,pv_input_voltage_1=NaN,battery_voltage_from_scc=NaN,battery_discharge_current=NaN
[2020-05-27 12:14:16] info serial:/dev/hidraw0:SEND query/device_rated_information
[2020-05-27 12:14:16] info serial:/dev/hidraw0:SEND_RAW [{"type":"Buffer","data":[81,80,73,82,73,248]},{"type":"Buffer","data":[84,13]}]
[2020-05-27 12:14:16] info serial:/dev/hidraw0:RECIVED_RAW Data: (NAKss Length: 7
[2020-05-27 12:14:16] info serial:/dev/hidraw0:RECIVED query/device_rated_information
[2020-05-27 12:14:16] info influx:SEND undefinedquery_device_rated_information grid_rating_voltage=NaN,grid_rating_current=NaN,ac_output_rating_voltage=NaN,ac_output_rating_frecuency=NaN,ac_output_rating_current=NaN,ac_output_rating_apparent_power=NaN,ac_output_rating_active_power=NaN,battery_rating_voltage=NaN,battery_recharge_voltage=NaN,battery_under_voltage=NaN,battery_bulk_voltage=NaN,battery_float_voltage=NaN,battery_type=NaN,current_max_ac_charging=NaN,current_max_charging_current=NaN,input_voltage_range=NaN,output_source_priority=NaN,charger_source_priority=NaN,parallel_max_num=NaN,machine_type=NaN,topology=NaN,output_mode=NaN,battery_re_discharge_voltage=NaN,pv_condition=NaN,pv_power_balance=NaN

on pip5048MS it works perfectly

any suggest?

cybertza, 2020-05-27:
RECIVED_RAW Data: (NAKss Length: 6)

Means bad command received by inverter, is there not another folder for the correct inverter, otherwise you would have to make one after finding the protocol manual.

Good news is the inverter is getting data, and replying, bad new is that it doesn’t know what you asking it. Aka No Ack or NAK

cybertza, 2020-05-27:
Different folders are different questions for different inverter models, and configurations

Tolmino, 2020-05-27:
yes, i know that there is something different....
but https://github.com/jblance/mpp-solar that is a python program created for pip4048MS reads PIP5048MK too without a new query, giving an unknow parameter in responce. 
so i have tried to add an unknown line to query, but without success

Tolmino, 2020-05-28:
https://www.fve-mp.cz/data/blob/product-application_pdf-20190724092401-8519-pip-gk-mk-protocol.pdf

Tolmino, 2020-05-28:
hi, i have found serial protocol comunication of mk series.
and if i add the missing parameter to retrieve in qpiri request i haven't a good response too

daromer2, 2020-11-29:
Closing. You just need to verify that CRC and if needed is correct including the query. Then you need to verify so that the software know how to retrieve it

closed 2020-11-29 by daromer2